PlanSys2/ros2_planning_system #266: Error executing a plan

MatteoLupinacci, 2023-07-18:
Hy, 

I'm trying to run a plan generated by a planner outside plansys2, but I always get the same error:

[ERROR] [executor_node-4]: process has died [pid 2604, exit code -11, cmd '/opt/ros/humble/lib/plansys2_executor/executor_node --ros-args -r __node:=executor -r __ns: =/ --params-file /tmp/launch_params_y660_zry --params-file /opt/ros/humble/share/plansys2_bringup/params/plansys2_params.yaml'].

There are no problems with domain files or setting up instances from the terminal. The very strange thing is that by trying to run the run plan-file <planFile> command several times, there are cases in which the execution starts and is successful. What can I do to solve this problem?

fmrico, 2023-07-19:
Hi @MatteoLupinacci 

May you give me more details to reproduce this problem (distro, domain, problem,...)?

MatteoLupinacci, 2023-07-19:
Thanks for the reply. As you can see from the error code I'm using ros humble. 
[ros2_ws.zip](https://github.com/PlanSys2/ros2_planning_system/files/12092414/ros2_ws.zip)

This is the workspace containing all files. In the `launch` folder there is the file with the commands to give to the plansys2 terminal in order to setting up the instances. The `plan.txt file` contains the plan, while the domain file is located in the pddl folder. I'm using 

> run plan-file plan.txt
in the plansys2 terminal to run the plan.

closed 2023-07-19 by MatteoLupinacci

fmrico, 2023-07-20:
Hi @MatteoLupinacci 

I have found your problem. In all your ids in the domain, change `-` by `_`

It seems that there is a bug that considers, for example, `is-free` an arithmetic expression `is` minus `free`. 

Best
Francisco

MatteoLupinacci, 2023-07-20:
Thanks, it works well now.