PlanSys2/ros2_planning_system #273: Add code auto-formatting to repo

sea-bass, 2023-09-16:
This is something I'm happy to work on by adding in [pre-commit](https://pre-commit.com/) plus a CI action to check auto-formatting.

However, before I try this I wanted to get the maintainers' go-ahead.

fmrico, 2023-09-21:
Hi @sea-bass 

CI is already passing the usual tests, like cpplint and uncrustify. If it does the same? What more does your proposal contribute to CI? Another concern is to have the opportunity for contributors to pass the same repo CI tests in their machines.

fmrico, 2023-09-21:
Is this related to #248?

sea-bass, 2023-09-22:
Yes it is, and didn't realize there was already a closed PR. My apologies.

I like pre-commit because it extends beyond just C++ code -- we use it to format Python, YAML, Dockerfiles, markdown docs, etc. And it also supports running `clang-format` on C++.

cpplint and uncrustify are a little more work to run, and most contributors (like myself) don't realize it's even there in CI until I fail some checks.

The `CONTRIBUTING.md` should definitely mention how to run these locally as part of commit requirements. Just a suggestion.

robodrome, 2023-10-05:
> Yes it is, and didn't realize there was already a closed PR. My apologies.
>
> I like pre-commit because it extends beyond just C++ code -- we use it to format Python, YAML, Dockerfiles, markdown docs, [...]

I agree. I had some issues with formatting.

fmrico, 2026-10-23:
Hi @sea-bass, thanks for raising this, and sorry it took so long. The CI already runs the `ament_lint` checks, so rather than adding pre-commit we document how to run them locally, with `uncrustify` fixing the format: #516. Closing.

closed 2026-10-23 by fmrico