PlanSys2/ros2_planning_system #280: `ProblemExpert::addInstance` is case-sensitive, but `ProblemExpert::addPredicate` is not

adlarkin, 2023-11-14:
Consider the following code:

```c++
const auto domain_test_str = R"(
(define (domain simple)
(:requirements :strips :typing :adl :fluents)

(:types
    Robot
    Location
    Object
)

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
; Predicates, Functions, and Actions can be defined below ... ;
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

)
)";

auto domain_expert = std::make_shared<plansys2::DomainExpert>(domain_test_str);

plansys2::ProblemExpert problem_expert(domain_expert);

problem_expert.addInstance(plansys2::Instance("rob", "Robot"));
problem_expert.addInstance(plansys2::Instance("table", "Location"));
problem_expert.addInstance(plansys2::Instance("apple", "Object"));
```

The `problem_expert.addInstance` calls fail, because [the domain is converted to all lowercase](https://github.com/PlanSys2/ros2_planning_system/blob/master/plansys2_domain_expert/src/plansys2_domain_expert/DomainReader.cpp#L47), but [`ProblemExpert::addInstance` does not convert the instance type to lowercase](https://github.com/PlanSys2/ros2_planning_system/blob/master/plansys2_problem_expert/src/plansys2_problem_expert/ProblemExpert.cpp#L402) in the method implementation.

If the `DomainExpert` is converting everything to lowercase, then other API calls that use the domain should also convert inputs to lowercase. This is being done when calling [`ProblemExpert::addPredicate`](https://github.com/PlanSys2/ros2_planning_system/blob/master/plansys2_problem_expert/src/plansys2_problem_expert/ProblemExpert.cpp#L461), and `ProblemExpert::addInstance` should follow the same pattern.

fmrico, 2023-11-16:
uuuuh, it seems like a bug that is easy to fix. Simply convert to lowercase in the `ProblemExpertNode`, probably.

Will you give it a try, or will you wait until I have time for this? I can review a PR to fix this if you feel strong ;)

adlarkin, 2023-11-16:
I opened a PR proposing a fix: https://github.com/PlanSys2/ros2_planning_system/pull/281

We can move the discussion there, thanks!

PR #281 (2023-11-19): Make `ProblemExpert` case-insensitive to Instance types
Closes #280 

My interpretation here is that Instance _names_ should be case-sensitive, but instance _types_ shouldn't be.
closed 2023-11-19 by fmrico