PlanSys2/ros2_planning_system #282: Multi-robot with inheritance from BtActionNode

robodrome, 2023-11-18:
I am trying to control two robots (e.g. `r2d2` and `3cpo`). I have a `MoveToGoal` action which inherits from `BtActionNode`. It works fine with one robot, but I need to hardcode the ROS2 namespace of the robot (e.g. `/r2d2`) during class instance construction. I tried `specialized_arguments` but that didn't work. I found some examples here:
- https://github.com/PlanSys2/ros2_planning_system/issues/221
- [test3](https://github.com/PlanSys2/ros2_planning_system/tree/master/plansys2_tests/test_3)
- [BT actions documentation](https://github.com/PlanSys2/ros2_planning_system/tree/master/plansys2_bt_actions)
- [PlanSys2 code related to specialized_arguments](https://github.com/search?q=org%3APlanSys2%20specialized_arguments&type=code)
- [PlanSys2 issues related to specialized_arguments](https://github.com/search?q=org%3APlanSys2+specialized_arguments&type=issues)
- [PlanSys2 documentation](https://plansys2.github.io/design/index.html?highlight=specialized_arguments#id16)

**QUESTION:** considering the setup below, how do I use one class like `MoveToWaypoint` with robots that are namespaced in ROS2?

### MoveToGoal.cpp
```cpp
namespace MyNamespace
{
MoveToGoal::MoveToGoal()
{
}

BT::PortsList MoveToGoal::providedPorts()
{
}

BT::NodeStatus MoveToGoal::on_tick()
{
}
}  // namespace MyNamespace

#include "behaviortree_cpp_v3/bt_factory.h"
BT_REGISTER_NODES(factory)
{
  BT::NodeBuilder builder = [](const std::string& name, const BT::NodeConfiguration& config) {
    return std::make_unique<MyNamespace::MoveToGoal>(name, "/r2d2/waypoint", config);
  };

  factory.registerBuilder<MyNamespace::MoveToGoal>("MoveToGoal", builder);
}
```

**NOTE:** the `/r2d2/waypoint` action name with `r2d2` prefix. I want the `r2d2` part to be set at launch, not hardcoded here. That way I can control multiple robots with one class.

### plansys2_myexample_launch.py

```python
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

def generate_launch_description():
    # Get the launch directory
    myexample_dir = get_package_share_directory('plansys2_myexample')
    namespace = LaunchConfiguration('namespace')

    declare_namespace_cmd = DeclareLaunchArgument(
        'namespace',
        default_value='',
        description='Namespace')

    plansys2_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(
            get_package_share_directory('plansys2_bringup'),
            'launch',
            'plansys2_bringup_launch_distributed.py')),
        launch_arguments={
            'namespace': namespace,
            'model_file': myexample_dir + '/pddl/domain_movetogoal.pddl',
            'problem_file': myexample_dir + '/pddl/problem_movetogoal.pddl',
            'namespace': namespace
        }.items()
    )

    # Specify the actions
    move_to_goal_unit_1_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_to_goal_unit_1',
        namespace=namespace,
        output='screen',
        parameters=[
            myexample_dir + '/config/params.yaml',
            {
                'action_name': 'move_to_goal',
                'enable_groot_monitoring': False,
                'bt_xml_file': myexample_dir + '/btxml/move_to_goal.xml'
            }
        ]
    )

    ld = LaunchDescription()

    ld.add_action(declare_namespace_cmd)

    # Declare the launch options
    ld.add_action(plansys2_cmd)
    ld.add_action(move_to_goal_unit_1_cmd)

    return ld
```

**NOTE:** I have tried `remappings` and `specialized_arguments`. I couldn't get it to work.
**NOTE2:** I have alterered ```plansys2_bringup_launch_distributed.py``` to take separate problem PDDL file.

### MoveToGoal.xml

```xml
<root main_tree_to_execute="MainTree">
        <BehaviorTree ID="MainTree">
                <Sequence name="root_sequence">
                        <MoveToGoal name="move_to_goal" unit="{arg0}" goal="${arg1}" />
                </Sequence>
        </BehaviorTree>
</root>
```

### params.yaml

```yaml
move_to_goal_unit_1:
  ros__parameters:
    plugins:
      - plansys2_move_to_goal_btnode
    waypoints: ["kitchen", "bedroom"]
    waypoint_coords:
      kitchen: [1, 2, 0]
      bedroom: [2, 3, 0]
```

robodrome, 2023-11-20:
**UPDATE 20231120:** 
I found a solution. It is not perfect, but it works; change `bt_xml_file /move_to_goal.xml` like this:

```xml
<root main_tree_to_execute="MainTree">
        <BehaviorTree ID="MainTree">
                <Sequence name="root_sequence">
                        <MoveToGoal name="move_to_goal" unit="{arg0}" goal="${arg1}" server_name="/${arg0}/waypoint/" >
                </Sequence>
        </BehaviorTree>
</root>
```

**NOTE:** `$arg0` (unit/robot) should be **identical** to the namespace of that robot. Using `server_name` fixes my issue for now. If there is a better way, please let me know. 

**SUGGESTION:** Perhaps this solution could be mentioned in the documentation?

closed 2023-11-20 by robodrome

fmrico, 2023-11-20:
I did some experiments some weeks ago, and I found this problem. I fix it, but I had no time to make a PR, but I have to do it.