PlanSys2/ros2_planning_system #290: Colcon build error on fresh install complaining about Behaviour Tree package

DigitalGabriele, 2024-02-07:
Hi,

I have Ubuntu 22 with ROS2. I followed the tutorial on how to install PlanSys2:

```
cd <your_workspace>
git clone -b <ros2-distro>-devel https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples.git src
colcon build --symlink-install
```

It failed on the colcon build with the following output:

```
Starting >>> plansys2_executor
--- stderr: plansys2_executor                   
CMake Error at CMakeLists.txt:20 (find_package):
  By not providing "Findbehaviortree_cpp_v3.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp_v3", but CMake did not find one.

  Could not find a package configuration file provided by
  "behaviortree_cpp_v3" with any of the following names:

    behaviortree_cpp_v3Config.cmake
    behaviortree_cpp_v3-config.cmake

  Add the installation prefix of "behaviortree_cpp_v3" to CMAKE_PREFIX_PATH
  or set "behaviortree_cpp_v3_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp_v3" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< plansys2_executor [4.00s, exited with code 1]
Aborted  <<< rosbag2_transport [7min 58s]      
```
I cloned the humble-devel branch from the following link: https://github.com/BehaviorTree/BehaviorTree.ROS2

I then removed build, install and log folders and started a fresh build of everything and I am still getting the above error.

fmrico, 2024-02-09:
Hi @DigitalGabriele 

PlanSys2 still uses BehaviorTreeCPP 3.x . Remove the version 4.x that you installed. Use apt to install the correct version of behavior tree. As far as I remember, it is `sudo apt install ros-humble-behaviortree-cpp-v3`. It also could be fixed using rosdep with your workspace.

Francisco.

DigitalGabriele, 2024-02-09:
Hi,

It says its already installed when I ran `sudo apt install ros-humble-behaviortree-cpp-v3 `but it doesnt work. 

```
ros-humble-behaviortree-cpp-v3 is already the newest version (3.8.6-1jammy.20240126.102626).
ros-humble-behaviortree-cpp-v3 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
```

fmrico, 2024-02-09:
Did you sourced the underlay before building?

Is the package in the list when you execute `ros2 pkg list|grep behavior"?

robodrome, 2024-02-19:
Try this:

```bash
git clone -b rolling https://github.com/PlanSys2/ros2_planning_system.git
git clone -b rolling https://github.com/PlanSys2/ros2_planning_system_examples
```

fmrico, 2024-02-23:
@DigitalGabriele Did these answers help you? Can we close the issue?

gemaRincon, 2024-03-20:
Hi! I have the same problem, when I do the colcon build I get an error in behaviortree_cpp. 

--- stderr: plansys2_executor
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findbehaviortree_cpp.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp", but CMake did not find one.

  Could not find a package configuration file provided by "behaviortree_cpp"
  with any of the following names:

    behaviortree_cppConfig.cmake
    behaviortree_cpp-config.cmake

  Add the installation prefix of "behaviortree_cpp" to CMAKE_PREFIX_PATH or
  set "behaviortree_cpp_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< plansys2_executor [8.91s, exited with code 1]

I am using Ros2 version foxy and I have done the steps you indicated and as you can see the behaviortree_cpp_v3 package is installed. 
:~/ros2_ws/prueba$ ros2 pkg list|grep behavior
behaviortree_cpp_v3

closed 2024-05-14 by DigitalGabriele