PlanSys2/ros2_planning_system #308: Is it possible to modify the domain via the DomainExpertClient?

Rezenders, 2024-05-06:
Hello,

Is it possible to modify the domain formulation via the DomainExpertClient?  For example, adding a precondition to an action.

Similar to how the problem formulation is modified with the ProblemExpertoClient? E.g.,  `problem_expert_->addPredicate(plansys2::Predicate("(robot_at r2d2 wp1)"));`

By checking the code, I concluded that this is not possible, but I would like to double-check with you.

Best,
Gustavo.

fmrico, 2024-05-07:
Well, the DomainExpert is currently static, but that is not necessarily written on Stone. It could be a good contribution if you want to work on It 😉

Rezenders, 2024-05-07:
Good to know. This is something I would like to work on.
I will investigate the code more, sketch a solution, and post it here before implementing something.