PlanSys2/ros2_planning_system #311: Implement rest of the clients in python

AravindaDP, 2024-05-30:
While implementing clients in python is discussed in https://github.com/PlanSys2/ros2_planning_system/issues/3 it seems only action performers in python is implemented with https://github.com/PlanSys2/ros2_planning_system/pull/263

Is there any plans in developing other clients like DomainExpertClient, PlannerClient, ProblemExpertClient and ExecutorClient as well?

I'm currently getting familiar with the project and would like to give it a try porting the rest of clients if it is a welcome contribution. Any directions I should be aware of? (Similar to guidelines in https://github.com/PlanSys2/ros2_planning_system/issues/3#issuecomment-655304135_)

fmrico, 2024-06-10:
There are no plans, but it could be a really good contribution :)

fmrico, 2026-10-09:
The Python clients requested here (`DomainExpertClient`, `ProblemExpertClient`, `PlannerClient` and `ExecutorClient`) are now available in `plansys2_support_py`. They were added in #383, improved in #387, #388 and #395, and have shipped since release 3.0.2, so client applications can now be written fully in Python. Closing as implemented; please open a new issue for any problem you find with them.

Thanks a lot @AravindaDP for asking for this, it is finally there.

closed 2026-10-09 by fmrico