PlanSys2/ros2_planning_system #326: Can "or" precondition allow more than 2 parameters

ipa-rwu, 2024-09-30:
For example I create an action as

```pddl
    (:durative-action move-wp
        :parameters (?w - workpiece ?cur ?nxt - position)
        :duration (= ?duration 1)
        :condition (and
            (at start (at ?w ?cur))
            (at start (clear ?nxt))
            (at start (or
                (and (= ?cur p1) (= ?nxt p2))  ;; Move from P1 to P2
                (and (= ?cur p2) (= ?nxt p3)) ;; Move from P2 to P3
                (and (= ?cur p3) (= ?nxt p4))
            ))
        )
        :effect (and
            (at end (not (at ?w ?cur)))
            (at end (at ?w ?nxt))
        )
    )
```

It seems that "or" only accept 2 paramters.
https://github.com/PlanSys2/ros2_planning_system/blob/rolling/plansys2_pddl_parser/include/plansys2_pddl_parser/Or.h#L14-L17

```cpp
class Or : public Condition {

```cpp
public:
	Condition *first, *second;

	Or()
		: first( 0 ), second( 0 ) {}
```

Is it possible to accept more than 2?

fmrico, 2024-10-02:
Try to expand second operator in a new `or` expression

fmrico, 2024-12-02:
Hi @ipa-rwu, did you manage to solve your problem?

I will close this issue. Please just go ahead and reopen if you still have problems.

closed 2024-12-02 by fmrico