PlanSys2/ros2_planning_system #328: Inefficient existential preconditions resolution

Rezenders, 2024-10-04:
Hi @fmrico,

Unfortunately, the solution I proposed for existential predicates is inefficient. It is literally exploring the whole state space to check if the predicates are satisfied. This can easily explode. For example, if there are 75 instances and an existential precondition has 5 variables, this results in 75^5 possible combinations to check, which is unfeasible. 

A concrete example (complete formulation can be found [here](https://github.com/Rezenders/suave_planning/tree/main/pddl)): 
```pddl
	(:derived (inferred-Fg_status ?fg ?IN_ERROR_NFR_string)
		(and
			(= ?IN_ERROR_NFR_string IN_ERROR_NFR_string)
			(exists (?mqa ?eqa ?mqav ?fd ?eqav)
 				(and
					(inferred-FunctionGrounding ?fg)
					(inferred-HasQAvalue ?fg ?mqa)
					(inferred-IsQAtype ?eqa water_visibility)
					(inferred-HasValue ?mqa ?mqav)
					(inferred-TypeFD ?fg ?fd)
					(inferred-HasValue ?eqa ?eqav)
					(inferred-HasQAestimation ?fd ?eqa)
					(inferred-IsQAtype ?mqa water_visibility)
					(inferred-FunctionDesign ?fd)
					(inferred-QAvalue ?mqa)
					(inferred-QAvalue ?eqa)
					(inferred-LessThan ?mqav ?eqav)
				)
			)
 		)
 	)
```

We need to come up with a mechanism to prune the possible variable combinations. I don't have any good ideas so far.

tobiaswjohn, 2024-10-07:
The general procedure should be to start from the state, i.e. known facts and build the set of all derived facts from that. This applies to the existential quantification and the derived predicates.

I assume, that there is a set `S` that contains all atomic formulas of the form `(p a_1 ... a_n)` that are true in the current state

- notations:
    - variables are named `?x_1` until `?x_n`
    - object names are `a_1` until `a_n` 
    - derived predicates have the form `(:derived (p ?x_1 ... ?x_n) P')`, where the predicate `(p ?x_1 ... ?x_n)` is derived, if the condition `P'` is satisfied (where `?x_1,...,?x_n` occur freely in `P'`)

## Algorithm for derived predicates
- The general semantics of derived predicates is defined in [this paper](https://www.jair.org/index.php/jair/article/view/10427) (see page 525 and the algorithm on page 526). The algorithm here is a version of that. In general, the algorithm tries to iteratively apply all rules (function `derive`) until no additional new facts can be derived.

- algorithm to compute all derived predicates from a state `S`, i.e. a set of satisfied atomic formulas
```
allDerived(S):
   S' := union(S, derive(S))
   while (S != S'):
      S := S'
      S' := union(S, derive(S))
   return S
```

- one single inference step for the derived predicates
```
derive(S):
   D := Set()
   forAll derived predicates derPred:
         case derPred = (:derived (p ?x_1 ... ?x_n) P') :
            E := eval(P', S)
            Derived := {(p a_1 ... a_n) | (x_1=a_1,...,x_n=a_n) in E}
            // CAVE: if some variables are never used in P': add all combinations of them to Derived
            D := union(D, Derived)
   return D
```

- the function `eval` returns for a piece of PDDL code (e.g. predicate, existential formula) the possible assignments that hold (given a state)
- variables that do not occur in the assignment are not restricted by the formula → can be assigned any object
- i.e. it maps PDDL code `P`and state `S`to a set
```
{{?x_1=a_1,...,?x_n=a_n}, {?x_1=a_1',...,?x_n=a_n'},...}
```

```
eval(P, S):
   case P = (p ?x_1 ... ?x_n) :
      return {{?x_1=a_1,...,?x_n=a_n} | (p a_1 ... a_n) in S}

   case P = (exists (?x_j) P') :
      E := eval(P', S)
      return {{?x_1=a_1,...,?x_n=a_n} | {?x_1=a_1,...,?x_n=a_n, ?x_j=a_j} in E}

   case P = (and (P' R')) :
      E_1 := eval(P', S)
      E_2 := eval(R', S)
      return join(E_1, E_2)   // join via shared variables
```

## Algorithm to decide preconditions

Based on the before mentioned, we can decide, if the precondition of an action is satisfied.
- general algorithm to compute, if preconditions are satisfied: compute all derived predicates → evaluate preconditions based on derived predicates
- algorithm: returns `true` if precondition is satisfied
- assumption: precondition is something that can be used in `eval`function
  ```
    evalPrecond(pre):
     S := state
     S_D := allDerived(S)
     if {?x_1=a_1,...,?x_n=a_n} in eval(pre, S_D): // CAVE: if eval-function does not set all variables --> those can have any value
        return true
     else:
        return false
  ```
- note: one could make this more efficient by feeding the already assigned variables to the `eval` function, but this would make the pseudocode more ugly

Rezenders, 2024-10-09:
@tobiaswjohn Thanks! I am working on improving the algorithm with your input. I will come back with some questions soon :)

tobiaswjohn, 2024-10-29:
@Rezenders I see you did quite some work and made good progress on this issue. You made quite some comments in the pull request #330 
Are there any specific points, where my input would be helpful?

Also: does the evaluation really take more than 30s on the (rather simple) example? This still sounds like quite some time :/

Rezenders, 2024-10-29:
> Also: does the evaluation really take more than 30s on the (rather simple) example? This still sounds like quite some time 

Yes, it takes around 37s to solve all derived predicates in the [domain_derived.pddl](https://github.com/Rezenders/ros2_planning_system/blob/new_evaluation/plansys2_problem_expert/test/pddl/domain_derived.pddl). However, I don't think this is a problem necessarily related to the new evaluation implementation, using only the SymK planner to find a plan using the previous domain and the [problem_derived.pddl](https://github.com/Rezenders/ros2_planning_system/blob/new_evaluation/plansys2_problem_expert/test/pddl/problem_derived.pddl) problem formulation, takes around 278 seconds (ps: you need to remove the fluents from the pddl formulation to make it work with symk).

```
ros2 run symk_ros fast-downward.py domain_derived.pddl problem_derived.pddl --search "sym_bd()"
INFO     planner time limit: None
INFO     planner memory limit: None

INFO     Running translator.
INFO     translator stdin: None
INFO     translator time limit: None
INFO     translator memory limit: None
INFO     translator command line string: /usr/bin/python3 /home/gus/symk_ros_ws/install/symk_ros/share/symk_ros/builds/release/bin/translate/translate.py domain_derived.pddl problem_derived.pddl --sas-file output.sas
Parsing...
Parsing: [0.000s CPU, 0.002s wall-clock]
Normalizing task... [0.000s CPU, 0.000s wall-clock]
Instantiating...
Generating Datalog program... [0.000s CPU, 0.000s wall-clock]
Normalizing Datalog program...
Normalizing Datalog program: [0.000s CPU, 0.002s wall-clock]
Preparing model... [0.000s CPU, 0.001s wall-clock]
Generated 67 rules.
Computing model... [59.000s CPU, 59.011s wall-clock]
3221937 relevant atoms
9022900 auxiliary atoms
12244837 final queue length
12889169 total queue pushes
Completing instantiation... [42.910s CPU, 42.940s wall-clock]
Instantiating: [104.750s CPU, 104.793s wall-clock]
Computing fact groups...
Finding invariants...
9 initial candidates
Finding invariants: [0.830s CPU, 0.835s wall-clock]
Checking invariant weight... [0.000s CPU, 0.000s wall-clock]
Instantiating groups... [0.000s CPU, 0.000s wall-clock]
Collecting mutex groups... [0.610s CPU, 0.594s wall-clock]
Choosing groups...
1288718 uncovered facts
Choosing groups: [8.050s CPU, 8.059s wall-clock]
Building translation key... [3.240s CPU, 3.241s wall-clock]
Computing fact groups: [26.610s CPU, 26.611s wall-clock]
Building STRIPS to SAS dictionary... [1.450s CPU, 1.457s wall-clock]
Building dictionary for full mutex groups... [9.400s CPU, 9.400s wall-clock]
Building mutex information...
Building mutex information: [1.280s CPU, 1.281s wall-clock]
Translating task...
Processing axioms...
Simplifying axioms... [6.330s CPU, 6.330s wall-clock]
Translator axioms removed by simplifying: 0
Processing axioms: [46.670s CPU, 46.680s wall-clock]
Translating task: [89.590s CPU, 89.637s wall-clock]
0 effect conditions simplified
0 implied preconditions added
Detecting unreachable propositions...
0 operators removed
0 axioms removed
74 propositions removed
Detecting unreachable propositions: [31.440s CPU, 31.461s wall-clock]
Reordering and filtering variables...
9 of 1288681 variables necessary.
0 of 0 mutex groups necessary.
1 of 644332 operators necessary.
8 of 644349 axiom rules necessary.
Reordering and filtering variables: [8.880s CPU, 8.895s wall-clock]
Translator variables: 9
Translator derived variables: 8
Translator facts: 18
Translator goal facts: 1
Translator mutex groups: 0
[... 82 lines cut ...]
party gustavo lucas albert einstein edson turtlebot rob2 livingroom (1)
[t=0.254721s, 435820 KB] Plan length: 1 step(s).
[t=0.254721s, 435820 KB] Plan cost: 1
[t=0.254721s, 435820 KB] Search time: 0.000432s
[t=0.254721s, 435820 KB] Total time: 0.254721s
Solution found.
Peak memory: 435820 KB
Remove intermediate file output.sas
search exit code: 0

INFO     Planner time: 278.27s

```

I don't think this is a particularly easy example due to the `(inferred-party ?p ?p2 ?p3 ?p4 ?p5 ?r ?r2 ?room)` derived predicate.  It contains 7 variables with the `object` type, i.e., allowing all instances of the formulation to be used to solve it. In addition, the body of the derived predicate doesn't put many restrictions on combining the variables. This results in `644332` possible combinations. 

The two main bottlenecks in the current approach are:

- Joining the variables sets: `return join(E_1, E_2, ..., E_N)   // join via shared variables`
Which is done here: https://github.com/Rezenders/ros2_planning_system/blob/e7798ce08458d12bdbbaa372407c80e1c509ce6d/plansys2_problem_expert/src/plansys2_problem_expert/Utils.cpp#L533-L610

- Grounding the derived predicates: `D := union(D, Derived)`
which is done here: https://github.com/Rezenders/ros2_planning_system/blob/e7798ce08458d12bdbbaa372407c80e1c509ce6d/plansys2_problem_expert/src/plansys2_problem_expert/ProblemExpert.cpp#L94-L156

@tobiaswjohn Fortunately, for our use case this is much faster, it takes around 1s. It is not ideal, but ....

tobiaswjohn, 2024-10-30:
Thank you for your comments. I think you made good progress and the time seems to be somehow usable for our use case (at least if we assume that executing an action takes several seconds anyways). At least, it should be small enough, that we can argue that it is not the center of our work to optimize this.
Also: @Rezenders  is the 1s with the current domains, that contain a lot of "useless" derivation rules? I guess, we could get rid of most of the derived predicates, if I would spend the effort to optimize our translation tool, which should decrease the run time.

One thing I was wondering about, which could be an optimization: Should one maybe have a list of all predicates, for which no assignment exists? E.g. for our work, consider the example from the top:
```
(:derived (inferred-Fg_status ?fg ?IN_ERROR_NFR_string)
		(and
			(= ?IN_ERROR_NFR_string IN_ERROR_NFR_string)
			(exists (?mqa ?eqa ?mqav ?fd ?eqav)
 				(and
					(inferred-FunctionGrounding ?fg)
					(inferred-HasQAvalue ?fg ?mqa)
					(inferred-IsQAtype ?eqa water_visibility)
					(inferred-HasValue ?mqa ?mqav)
					(inferred-TypeFD ?fg ?fd)
					(inferred-HasValue ?eqa ?eqav)
					(inferred-HasQAestimation ?fd ?eqa)
					(inferred-IsQAtype ?mqa water_visibility)
					(inferred-FunctionDesign ?fd)
					(inferred-QAvalue ?mqa)
					(inferred-QAvalue ?eqa)
					(inferred-LessThan ?mqav ?eqav)
				)
			)
 		)
 	)
```
If we saved somewhere, that e.g. "inferred-FunctionDesign" is empty, i.e. no assertion with this predicate exists, then we don't need to evaluate any of the other predicates. I think, this could speed up the whole process quite a bit. 
Similarly, we could have a list of the predicates for which new assertions were added in the last iteration of constructing `D`, i.e. if there is no predicate for which the assertions were updated --> do not evaluate any of the predicates, as this will not lead to new assertions.
To summarize: we should keep track of the predicates for which no assertions exist and the one for which new assertions where added in the last iteration of the loop. If an `(and ...)` or an `(or ...)` occurs, check that there is no predicate in the "empty" list and at least one in the "updated" list before doing the evaluation.

Just a vague idea, that might be not too complicated to implement. :) But I think, we can also focus on other things than optimizing this algorithm, if we want.

Rezenders, 2024-11-05:
> Similarly, we could have a list of the predicates for which new assertions were added in the last iteration of constructing D, i.e. if there is no predicate for which the assertions were updated --> do not evaluate any of the predicates, as this will not lead to new assertions.
To summarize: we should keep track of the predicates for which no assertions exist and the one for which new assertions where added in the last iteration of the loop. If an (and ...) or an (or ...) occurs, check that there is no predicate in the "empty" list and at least one in the "updated" list before doing the evaluation.

In addition to this, when applying the effect of actions, we could keep track of the predicates inserted/removed and only reevaluate the derived predicates that could be impacted instead of reevaluating all derived predicates. Constantly reevaluating all derived predicates is a current bottleneck to the approach

Rezenders, 2024-11-07:
I had an idea: instead of saving the derived predicates as a vector, we could save it as a direct acyclic graph representing the dependencies between the derived predicates. For example:

The derived predicates in the following domain:
```pddl
(:derived (inferred-InstanceA ?a)
  (and
    (= (?a instanceA))
  )
)

(:derived (inferred-A ?a)
  (and
    (predicateA ?a)
  )
)

(:derived (inferred-B ?b)
  (and
    (predicateB ?b)
  )
)

(:derived (inferred-AA ?a)
  (and
    (inferred-A ?a)
  )
)

(:derived (inferred-BB ?b)
  (and
    (inferred-B ?b)
  )
)

(:derived (inferred-AB ?a ?b)
  (and
    (inferred-A ?a)
    (inferred-B ?b)
  )
)
```

Could be represented with the following graph:

[image]

Then, instead of the current algorithm for solving derived predicates:

```
solveAllDerived(S):
   S' := derive(S)
   while (S != S'):
      S := S'
      S' := derive(S)
   return S

derive(S):
  for all d=(:derived (p ?x_1 ... ?x_n) C):
    E := evaluate(C, S)
    Derived := ground(p, E)
    S := union(S, Derived)
  return S
```

We could traverse the graph and solve the derived predicates in the correct order. This only requires one pass to solve all derived predicates, in contrast to the previous approach, which could take several iterations to finish. Furthermore, most critically, the last iteration from the previous approach doesn't solve any additional derived predicate. It only confirms that all predicates that could be solved were solved.

The graph search could be either breadth-first or depth-first, I don't think it makes any difference.

### Apply effects
With the graph representation, when applying the effects of an action in a given state, we can check the graph to find out which derived predicates the effect impacts and only solve these derived predicates again. In the example above, if instances of `predicateA` are added or removed we would only need to solve `inferred-A`, `inferred-AA`, and `inferred-AB`.

### Prune derived
Furthermore, since the end goal of solving the derived predicates is to use them to evaluate whether the actions being executed have their preconditions satisfied at execution time and to generate a plan-graph from the action plan, we could prune the derived predicate graph to contain only the derived predicates required for the actions in the current plan. For example:

Given the following actions and the pddl formulation above:
```pddl
(:action actionA
  :parameters (?a)
  :precondition (and
    (inferred-A ?a)
  )
  :effect (and
    (predicateB ?a)
  )
)

(:action actionB
  :parameters (?b)
  :precondition (and
    (inferred-b ?b)
  )
  :effect (and
    (goal ?b)
  )
)
```

Suppose the current plan is:
```
(actionA instance)
(actionB instance)
```

The derived predicates graph could be pruned to:

[image]

tobiaswjohn, 2024-11-07:
I like the idea with the graph! I think this an work well, as long as the set of grounded derived predicates is not too large. 
But it also looks like quite some effort to implement. :D

fmrico, 2024-11-08:
Graph way looks promising :)

Rezenders, 2025-06-08:
Hi @fmrico @tobiaswjohn ,

The approach I proposed above for solving the derived predicates in depth first order **only works for acyclic graphs**. In our application (SUAVE), the graph has some cyclic parts, see the [figure below](##SUAVE).

Thus, to solve this problem, I made some changes to the algorithm. The algorithm now:

1. First, it uses [Tarjan's algorithm](https://en.wikipedia.org/wiki/Tarjan%27s_strongly_connected_components_algorithm) to identify strongly connected components (SCCs), returned in reverse topological order;
2. Acyclic SCCs are evaluated in order;
3. Cyclic SCCs are resolved using a fixpoint algorithm, that is, all derived predicates in the SCC are iteratively evaluated until no further changes occur;

## Pseudocode:
```cpp
for (SCC in reverse_topological_order) {
  if (SCC is acyclic) {
    evaluate each derived predicate once;
  } else {
    do {
      changed = false;
      for (derived in SCC) {
        if (evaluate(derived) changes state) changed = true;
      }
    } while (changed);
  }
}
```

## Example

PDDL example:
```PDDL
(:derived (der1 ?x)
    (pred1 ?x))

  ;; Recursive trio (new SCC of size 3)
  (:derived (der2 ?x)
    (or (pred2 ?x) (der3 ?x)))

  (:derived (der3 ?x)
    (or (der4 ?x) (pred3 ?x)))

  (:derived (der4 ?x)
    (or (der2 ?x) (pred4 ?x)))

  ;; Recursive pair
  (:derived (der5 ?x)
    (or (der6 ?x) (pred2 ?x)))

  (:derived (der6 ?x)
    (der5 ?x))

  ;; Acyclic derived predicate depending on cyclic predicates
  (:derived (der7 ?x)
    (and (der2 ?x) (pred5 ?x)))

  ;; Independent acyclic derived predicate
  (:derived (der8 ?x)
    (pred3 ?x))

  ;; Connects to both cycles to eliminate disjoint subgraphs
  (:derived (der9 ?x)
    (and (der4 ?x) (der6 ?x)))
```

Graph:

[image]

Possible output of Tarjan's algorithm in topological order. The SCCs with 1 node are acyclic, the SCCs with more than 1 node are cyclic:

1. SCC₁: {der1}
2. SCC₂: {der8}
3. SCC₃: {der2, der3, der4} (can only be solved after der1, der8 are done)
4. SCC₄: {der5, der6}
5. SCC₅: {der7}
6. SCC₆: {der9}

Dependencies among SCCs:
 
- der7 depends on der2 (SCC₁)
- der9 depends on der4 (SCC₁) and der6 (SCC₂)
- der8 and der1 are leaves

SCC 1, 2, 5, 6 are simply evaluated once. SCC 3 and 4 are evaluated with the fixpoint algorithm

## Assumptions and Limitations
The proposed solution supports:

1. Both acyclic and cyclic dependencies between derived predicates,
2. Recursive axioms,
3. Axioms with conjunctions or disjunctions in their preconditions,
4. Negated axioms in preconditions, as long as the negated axiom is not in the same SCC.

**Limitation**:
Negated axioms in the precondition of another axiom within the same SCC cannot be handled by this approach.

A possible solution could be to check the paper [In defense of PDDL axioms](https://gki.informatik.uni-freiburg.de/papers/thiebaux-etal-aij-final.pdf) and check their algorithm/proof for compiling derived predicates and make sure our algorithm matches. AFAIK, their algorithm assumes the axioms' preconditions are in negated normal form (NNF), and they create a stratified set for the axioms, then solve it with a fix-point algorithm.

From my understanding, the described solution already fulfills these requirements for cyclic SCCs. To fully support the cases with negation in acyclic SCCs, it may be necessary to explicitly stratify the acyclic SCCs and apply the fixpoint algorithm accordingly. 

## SUAVE

[image]

## Implementation

I finished implementing all this. It seems to be working, but yesterday I made some changes that made the algorithm slow. I think I am copying some large vectors instead of passing them as references or moving them. I will work on this for now and I will update you when it is done.

tobiaswjohn, 2025-06-11:
I have two small comments on your algorithm:
1. I am not sure how your `evaluate(derived)` method works. But if run time is an issue, there might be potential to optimize it, e.g. by only considering the derived predicates in the SCC (because only those can change during the loop) or by using the dependencies to determine, which predicate can be effected by the change that was made by adding a derived predicate. I.e. not checking all derived predicates in the SCC, but only those that might have been affected by the last addition of a derived predicate. maybe you already do this, otherwise, there might be some time to gain there.
2. Negated axioms should not be an issue, they are not allowed to occur according to the [definition of PDDL 2.2](https://planning.wiki/_citedpapers/pddl222004.pdf). In the "defense of axioms" paper, the authors allow a restricted form of negation. In particular, axioms can NOT depend on negated variants of axioms that are within the same SCC (this is what "stratified" ensures). So maybe your algorithm also already works for such axioms.

Rezenders, 2025-06-14:
Hi @tobiaswjohn 

> I.e. not checking all derived predicates in the SCC, but only those that might have been affected by the last addition of a derived predicate.

This is partially done already. This is checked when an action is applied, i.e., I check which derived predicates need to be re-evaluated given an action's effects. But I guess this could also be done for predicates added at runtime via measurements.

> Negated axioms should not be an issue, they are not allowed to occur according to the [definition of PDDL 2.2](https://planning.wiki/_citedpapers/pddl222004.pdf). In the "defense of axioms" paper, [...]

Thanks for the clarification. I had understood it wrong. I thought there was a way to transform the sets before evaluating to make it "safe" to have negated conditions. 
Given your explanation, I believe our method satisfies this condition. The acyclic parts are always stratified, that is what the dependencies/edges guarantee, and the cyclic SCCs are not stratified.

Rezenders, 2025-06-14:
The evaluation time is not an issue anymore, at least for our use case.

After analyzing the ["SimpleBT" plan graph](https://arxiv.org/abs/2101.01964) algorithm, I came to the conclusion that this algorithm was designed to work under certain assumptions, which are not satisfied with our current model. Thus, it wouldn't be trivial/enough to modify/extend/patch it, we would need to design a new algorithm.
For the sake of time, given our deadline, I moved on with the approach of simply executing the plan in sequence. For that, I created a new [Sequential BT builder plugin](https://github.com/Rezenders/ros2_planning_system/blob/new_evaluation/plansys2_executor/include/plansys2_executor/bt_builder_plugins/simple_bt_builder.hpp) that generates a BT to execute the plan in sequence.

closed 2026-10-17 by fmrico