PlanSys2/ros2_planning_system #329: Errors parsing expressions

fmrico, 2024-10-17:
Hi!!

I am getting these errors, and I would like to know if it is really an error. 

```
getExpr: Error parsing expresion [(and (car_assembled car1))]
getExpr: Error parsing expresion [(car_assembled car1)]
getExpr: Error parsing expresion [(and (car_assembled car2))]
getExpr: Error parsing expresion [(car_assembled car2)]
getExpr: Error parsing expresion [(and (car_assembled car3))]
getExpr: Error parsing expresion [(car_assembled car3)]
```

https://github.com/PlanSys2/ros2_planning_system/blame/b82cc6fe5d577e481b0aa01b6b779f94d33b9ebe/plansys2_pddl_parser/src/plansys2_pddl_parser/Utils.cpp#L278

Thanks!!!

roveri-marco, 2024-10-17:
@fmrico How can I reproduce the problem? Can you circulate the PDDL domain file and the problem file or the commands to reproduce them?

Rezenders, 2024-10-28:
I don't believe this is an actual error.

In the following code, `getExpr(expr)` is being called to get the expression, and it is also being used to check if `expr` is an expression. When it is not an expression, it prints that there was an error parsing the expression, but it is not an error. `expr` isn't an expression. I believe the print can be removed

https://github.com/PlanSys2/ros2_planning_system/blob/b82cc6fe5d577e481b0aa01b6b779f94d33b9ebe/plansys2_pddl_parser/src/plansys2_pddl_parser/Utils.cpp#L196-L202