PlanSys2/ros2_planning_system #333: Jazzy debian package release

gregtkogut, 2024-11-08:
I don't know if this falls under the scope of the PlanSys2 maintainers, but is there a known time that ROS 2 jazzy packages will be released to `main`: https://repo.ros2.org/status_page/ros_jazzy_default.html?q=plansys2 ?

The immediate symptom is that `apt install ros2-jazzy-plansys2-core` won't work on the `osrf/ros:jazzy-desktop-full` docker image.

fmrico, 2024-11-08:
Ok, thanks. I have to check It. And yes, It is my responsability 😕

emanuelazfernandes, 2025-01-14:
@fmrico it seems like the problem in the failed build (action workflow) is related to the adding of rqt_gui_cpp/include/plugin.hpp when it used to be rqt_gui_cpp/include/plugin.h
From the commit history in the ros_visualization/rqt repo, it seem like this was added only for rolling and not for jazzy.
If it would be useful, I could create a PR but I don't think I have permission to trigger the action to verify and test.

fmrico, 2025-02-05:
> [@fmrico](https://github.com/fmrico) it seems like the problem in the failed build (action workflow) is related to the adding of rqt_gui_cpp/include/plugin.hpp when it used to be [...]

No, the problem was related to CI. I was checking out `rolling` to test the branch `jazzy-devel`. See https://github.com/PlanSys2/ros2_planning_system/commit/3fc9e946067c75169772851c5d762d323efd5383

closed 2025-02-05 by fmrico