PlanSys2/ros2_planning_system #343: plansys2 plans and executes multiple tasks simultaneously

hujf123, 2025-02-21:
Hello.
I want to use plansys2 as a task manager to manage multiple robots to perform different tasks. Can it plan and execute different plans and tasks at the same time? Does plansys2 only support single - thread plan execution?

SebastianGrey8, 2025-03-06:
Thank you for your response.
I want to implement a factory scenario that includes assembly and transportation, where these two types of tasks are independent, and the time required to complete an item's assembly or transportation varies. The factory currently has two robots, both capable of performing assembly and transportation tasks. The assembly and transportation tasks are randomly assigned based on factory orders. Currently, PlanSys2 can only execute one goal at a time, so it is not possible to simultaneously send multiple transportation goals and assembly goals.Running different PlanSys2 instances will not allow the two robots to effectively interleave and complete tasks.

Best

fmrico, 2025-03-13:
I think it is possible to do it without multiple PlanSys2 instances. Use the specialized arguments parameters for the action performers in each robot, and include the robot as first argument in all PDDL actions

See the factory assembly example

roveri-marco, 2025-05-22:
One additional comment. You can use one instance of plansys2 to control different robotic agents. Indeed, with PDDL you can model easily this, and then PlanSys2 allows for two or more actions to be executed in parallel provided there are actions implemented and as many instances as needed of each actions (indeed, if you expect two goto(r1, x1,y1) and goto(r2, x2, y2) actions to be executed in parallel by robot r1 and robot r2 to go to the respective pose, you need to instances of the action goto, otherwise the execution fails.

fmrico, 2026-10-09:
As discussed above, a single PlanSys2 instance can coordinate several robots. Model the robot as a parameter of every PDDL action, put all the orders into one goal, and run one action performer per robot, using `specialized_arguments` to bind each performer to its robot. The executor then runs non-conflicting actions of different robots in parallel. Closing as answered; please open a new issue if you find a concrete limitation with this approach.

Thank you @SebastianGrey8 for the question and for the discussion.

closed 2026-10-09 by fmrico