PlanSys2/ros2_planning_system #344: Adding BT creation support for forall and imply

eshan-savla, 2025-02-26:
Hi all,

I would like to make use of the _forall_ and _imply_ pddl keys for our use case and noticed that the appropriate BT creation functionalities are missing, and is also mentioned in this issue #314 . I would love to contribute but am a little lost on what exactly needs to be implemented. I have tried to study the program flow once the executor is triggered and have summarized my understanding of it below. If I have misunderstood something, please do correct me 😅. Additionally, I have a few questions and would greatly appreciate it, if somebody could help me out with them. 

- `execute` method from Executor Node handles retrieval of BT msgs requested from Domain expert and BT building.
- `getTree` method from Domain expert for both keys would need to be implemented. I have a few questions regarding this.
- Sub goal ordering occurs after action map creation, which also requests durative actions from domain expert. I presume only action effects are important here. 
- BT nodes for conditions, actions and effects are built. I would assume that the respective BT node creation methods for _forall_ and _imply_ would also need to be implemented. 

**Questions**:

1. What is the purpose of the _Lifted_ class? Is it relevant for forall or imply? I ask because it is being used in Ground conditions to seemingly access constants. 
2. I don't understand how the param values are set in the Tree msg. Using the Ground class as an example, I figured that based on the value in param vector, variable or constant values are set. However, what is to be understood when the param name is set to the param int value with "?" prepended to it?
3. Once the new msgs are created, where should the new BT Node functions be implemented? I haven't been able to spot where the BT Node implementations for the other types are. 

I aim to work on this for the humble branch first and would add these functionalities to other branches later. I would greatly appreciate any assistance here!

eshan-savla, 2025-03-03:
Should #345 be accepted, this can be closed.

fmrico, 2026-10-09:
This duplicates #314: both ask for `forall` (and `imply`) support when building the behavior trees, and the parser still rejects them with `UnsupportedConstruct`. The implementation you proposed is in #347, which needs a rebase on current `rolling` and an agreement on how to report constructs that a planner does not support. To keep the discussion in one place, we'll track the feature in #314 and the work in #347. Closing as duplicate.

Thank you @eshan-savla for both the issue and the PR, your implementation is the base we'll build on.

closed 2026-10-09 by fmrico