PlanSys2/ros2_planning_system #351: [BUG] Cannot change `/planner_client`  `plan_solver_timeout` parameter

Rezenders, 2025-03-17:
When running PlanSys2 with the following parameter file:

```yml
planner_client:
  ros__parameters:
    plan_solver_timeout: 60.0

planner:
  ros__parameters:
    plan_solver_timeout: 60.0
    plan_solver_plugins: ["OPTIC"]
    POPF:
      plugin: "plansys2/POPFPlanSolver"
    OPTIC:
      plugin: "plansys2/OPTICPlanSolver"

domain_expert:
  ros__parameters:
    validate_using_planner_node: false
```

The planner_client logs:
```
[krr_mirte_pddl-6] [INFO] [1742217591.677162029] [planner_client]: Planner CLient created with timeout 0
```

I run `ros2 param list` to check what is going on, and I get the following: 
```bash
$ ros2 param list
/domain_expert:
  POPF.arguments
  POPF.output_dir
  model_file
  use_sim_time
  validate_using_planner_node
Exception while calling service of node '/domain_expert_client': None
/executor:
  action_time_precision
  action_timeouts.actions
  bt_builder_plugin
  default_action_bt_xml_filename
  default_end_action_bt_xml_filename
  default_start_action_bt_xml_filename
  enable_dotgraph_legend
  print_graph
  use_sim_time
Exception while calling service of node '/executor_client': None
/krr_mirte_pddl_node:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
/move:
  action_name
  rate
  specialized_arguments
  use_sim_time
  wp0
  wp_bread
  wp_dishwasher
  wp_spoon_clean
  wp_spoon_dirty
  wp_tableware
  wp_trash
/pick:
  action_name
  rate
  specialized_arguments
  use_sim_time
/place:
  action_name
  rate
  specialized_arguments
  use_sim_time
/place_dishwasher:
  action_name
  rate
  specialized_arguments
  use_sim_time
/planner:
  OPTIC.arguments
  OPTIC.output_dir
  OPTIC.plugin
  plan_solver_plugins
  plan_solver_timeout
  use_sim_time
Exception while calling service of node '/planner_client': None
/problem_expert:
  model_file
  problem_file
  use_sim_time
Exception while calling service of node '/problem_expert_client': None
/use_dishwasher:
  action_name
  rate
  specialized_arguments
  use_sim_time
```

Note the error with the `domain_expert_client`, `executor_client`, `planner_client`, and `problem_expert_client`: 
```
Exception while calling service of node '/planner_client': None
```

This is the "PlanSys Controller" node I am running: https://github.com/kas-lab/krr_mirte_task_decision_layer/blob/main/krr_mirte_pddl/src/krr_mirte_pddl.cpp
I based this node on this example: https://github.com/PlanSys2/ros2_planning_system_examples/blob/rolling/plansys2_patrol_navigation_example/src/patrolling_controller_node.cpp

I am not sure what is going on.  Should I be spinning the node differently?  Is this a PlanSys2 bug? A rclcpp bug? A bug with the default DDS for ROS humble?

#### Setup

I have Ubuntu 22.04 with ROS 2 Humble and PlanSys2 https://github.com/PlanSys2/ros2_planning_system/commit/e2fd96eb26cd00b52416c918f63fff36704df5b2

#### Reproduce error

```Bash
xhost +
```

Terminal 1:
```Bash
docker run -it --rm -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro ghcr.io/kas-lab/krr_bringup:main ros2 launch mirte_gazebo gazebo_mirte_master_navigation.launch.py
```

Terminal 2:
```Bash
docker run -it --rm -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro ghcr.io/kas-lab/krr_bringup:main ros2 launch krr_mirte_pddl krr_mirte_pddl_launch.py
```

For this specific example in the docker image, PlanSys works without a problem since the planner returns before the timeout. But if you change the [plansys config file](https://github.com/kas-lab/krr_mirte_task_decision_layer/blob/main/krr_mirte_pddl/config/plansys2_params.yaml) (src/krr_mirte_task_decision_layer/krr_mirte_pddl/config/plansys2_params.yaml) to use the OPTIC planner instead of POPF, it doesn't work anymore since the OPTIC planner is taking more than 15 seconds to return.

fmrico, 2025-03-17:
Can you confirm this bug?

Rezenders, 2025-03-17:
@fmrico, this is not a bug. I was doing something wrong. Sorry for opening this issue.

I forgot to load the param file when launching my controller node. I was only loading it in the PlanSys bring-up.