PlanSys2/ros2_planning_system #380: Unable to install the package

RareCentipede, 2025-08-07:
Hi, I wanted to use PlanSys2 for my thesis but I cannot seem to install the package for ros2 humble. I'm running Ubuntu 22.04. Running the command ```sudo apt install ros-humble-plansys2-*``` I get the following errors:

```
sudo apt install ros-humble-plansys2-*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-plansys2-*
E: Couldn't find any package by glob 'ros-humble-plansys2-*'
```

I then cloned the humble-devel branch of the Plansys2 repo but then I was prompted with the error that `cascade_lifecycle` is missing. I solved this by cloning and building this [repo](https://github.com/fmrico/cascade_lifecycle), and now I am stuck because I am missing `ros-humble-popf` and apt cannot find it. I tried cloning the corresponding repo from fmrico but it does not work.

How can I solve this issue? Is it just not compatible in my Ubuntu/Ros2 distro? Thank you in advance.

fmrico, 2025-08-08:
Hi @RareCentipede 

There should be packages for Humble. I am far from my computer this week, but I will have a look at it early next week.

fmrico, 2025-08-13:
The packages should be created for Humble, as they are in the rosdistro: https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml

All the packages seem to be building correctly: https://repo.ros2.org/status_page/ros_humble_default.html

What is your Linux version and ROS 2 distro?

RareCentipede, 2025-08-13:
I am using Ubuntu 22.04 and Humble. Maybe I'm missing some repository? I'm not sure how to check or add them, apt still cannot find the package.

fmrico, 2026-10-09:
The PlanSys2 packages, as well as `popf` and `cascade_lifecycle`, are released for Humble (PlanSys2 2.0.9). `Unable to locate package` usually means that the ROS 2 apt repository is not configured, or that `sudo apt update` was not run after adding it. Please follow the ROS 2 Humble installation guide to set up the repository and then run `sudo apt install ros-humble-plansys2-*` again. Closing since there has been no activity; feel free to reopen if it still fails with the repository configured.

Thanks @RareCentipede for reporting it.

closed 2026-10-09 by fmrico